mechano/Kingroon_KP3S_Pro_Marlin_Firmware #10: Enable Host actions command

obones, 2024-01-08:
Hello,

I have just applied the latest ABL firmware to my printer and while the update worked just fine, octoprint is now reporting this:

> Info
> 
> Your printer's firmware supports host action commands, but they are disabled. The firmware will not inform OctoPrint about job pause & cancellations, filament runouts, and similar firmware-side events. If you want this functionality, you need to change your firmware's configuration.

The message gives a link to this message:
https://community.octoprint.org/t/octoprint-tells-me-my-firmware-lacks-support-for-host-action-commands-what-does-this-mean/34588

Could you thus do this in a next iteration of this firmware?

Many thanks for your help

mechano, 2024-01-08:
Ok, I'll add to all my firmwares.

Is it ok 115200 speed for USB-serial instead of 250000?

obones, 2024-01-08:
That's already the speed it's connecting at, so fine by me:

[image]

obones, 2024-01-08:
Thanks a lot!

closed 2024-01-08 by obones