mechano/Kingroon_KP3S_Pro_Marlin_Firmware #18: Printhead only moves up when trying to auto level (BLTouch)

SprocketTools, 2025-01-04:
I have a KP3S Pro that I've upgraded to use a Bambu clone hotend, BLTouch, PEI sheet, and the ABL 350 firmware.  After flashing the firmware, I used the "bed leveling" tool to level the bed.  After pressing this, the printhead moves to the middle of the printbed, then extends and retracts the BLTouch sensor 3 times before saying "home Z first."  From here I found out that it was impossible to make the Z axis move downward, as homing the Z axis or manually moving the Z axis downward resulted in no movement.  On the other hand, moving the printhead upwards is permitted.

This sounds like a sensor permanently in the "triggered" position, but my wiring configuration matches the instructions in #8 .  Not sure how to fix this issue.

Photos included for reference:

[image]
[image]

mechano, 2025-01-04:
Try moving the black/white plug on Z+

SprocketTools, 2025-01-04:
Interesting, I've seen so many people point out that the plug needs to go in the Z- port.  Placing it in the Z+ port has allowed it to work normally.  Thanks for everything (including this project)!

closed 2025-01-04 by SprocketTools

mechano, 2025-01-04:
I've left Z- free for people who want still to use Z- switch for other
purposes.