mechano/Kingroon_KP3S_Pro_Marlin_Firmware #19: Y axis motor skips during printing

SprocketTools, 2025-01-05:
Whenever the printer tries to move the printbed during printing, in most cases the Y axis motor will freeze and make rattling sounds, akin to if it was trying to exceed the physical limits of the axis.  It moves fine during calibration and manual movement, but once printing starts, it freezes at some point before the initial purge is completed.

Fixes tried so far:
- tightening Y axis belt: no effect
- disabling input shaping: no effect
- raising the Z axis offset slightly: slightly reduces frequency of lockups
- Increasing Y axis motor current: not feasible, as M906 and M907 return as "unknown commands" in Pronterface
The only solution I found so far is raising the Z axis offset to where the filament barely sticks to the printbed at all, which can't be used for actual printing.

This issue was not present at all with the stock firmware, as the printer could handle situations that required far more torque than what is currently needed to move the nozzle across the printbed.  The only modifications I have made to the Y axis prior to installing Marlin firmware is tightening the belt and leveling the printbed.

mechano, 2025-01-05:
Sorry but this firmware is used by thousands people around the world.
Every time I have discuss for failure they where always human errors.

SprocketTools, 2025-01-28:
Alright, update after nearly a month of trying to figure it out.

What I had not noticed until recently is that when the Z axis goes down to the printbed, a part of the Z axis frame was directly colliding with the edge of the printbed, which in turn caused enough friction to make the bed lock up.  Taking an angle grinder to the problem worked, and the printer is now working (with some tuning required).  

Thanks again!

closed 2025-01-28 by SprocketTools

mechano, 2025-01-28:
If you are talking of that part of the gantry on left side, I corrected with 3mm printed shims under the 16mm silicon bush under the bed instead of the springs.

SprocketTools, 2025-01-29:
It was here - due to my hotend being 5mm shorter this began to collide with the bed.
[image]

mechano, 2025-01-29:
Yes I had the same problem and had to put different spacers under the bed. 

As you can see they exists of different sizes. It works good with a size near 19-20mm

https://aliexpress.com/item/1005008063875180.html